riscv-software-src/opensbi #272: How to cross compile opensbi for RV64IMAFD ?

billasurendra, 2022-10-12:
Hello,

I have Linux kernel Image without compressed instruction support and rootfs also without c model. Now I am planning to cross compile opensbi . Can anyone please tell me how to cross compile opensbi without compressed instruction support.

Thanks
Billa

avpatel, 2022-10-12:
Try this:
`make PLATFORM=generic PLATFORM_RISCV_ISA=rv64imafd PLATFORM_RISCV_XLEN=64`

billasurendra, 2022-10-13:
> Try this:
> make PLATFORM=generic PLATFORM_RISCV_ISA=rv64imafd PLATFORM_RISCV_XLEN=64
>
> Dear Anup,
I Tried your suggestion, but I am getting following error. Can you please
help me out ?

Error:

 AR        platform/generic/lib/libplatsbi.a
 AS        platform/generic/firmware/payloads/test_head.o
 CC        platform/generic/firmware/payloads/test_main.o
 MERGE     platform/generic/firmware/payloads/test.o
 ELF       platform/generic/firmware/payloads/test.elf
/root/MDP/RV64G_Fedora_Distro/rv64g-os/cross-tools/lib/gcc/riscv64-unknown-linux-gnu/12.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld.bfd:
warning:
/root/MDP/RV64G_Fedora_Distro/packages/opensbi/build/platform/generic/firmware/payloads/test.elf
has a LOAD segment with RWX permissions
 OBJCOPY   platform/generic/firmware/payloads/test.bin
objcopy: Unable to recognise the format of the input file
`/root/MDP/RV64G_Fedora_Distro/packages/opensbi/build/platform/generic/firmware/payloads/test.elf'
make: *** [Makefile:534:
/root/MDP/RV64G_Fedora_Distro/packages/opensbi/build/platform/generic/firmware/payloads/test.bin]
Error 1

billasurendra, 2022-10-13:
FYI,

I am using following command

make PLATFORM=generic
FW_PAYLOAD_PATH=/root/MDP/RV64G_Fedora_Distro/packages/linux-5.19.2/arch/riscv/boot/Image
PLATFORM_RISCV_ISA=rv64imafd_zicsr_zifencei PLATFORM_RISCV_XLEN=64
PLATFORM_RISCV_ABI=lp64d

In the makefile ABI option only lp64 is available.

billasurendra, 2022-10-14:
After doing appropriate changes in Makefile, opensbi cross compiled for rv64imafd_zicsr_zifencei. Finally I have Linux kernel Image, Linux distro Image and opensbi are without compressed instruction support. Started booting my Linux distro image with opensbi but surprisingly  I could see rv64imafdc. Can anyone please help me to identify the reason why I am getting that line.
 
Result:

OpenSBI v1.1-61-gb6e520b
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : riscv-virtio,qemu
Platform Features         : medeleg
Platform HART Count       : 8
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 10000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : sifive_test
Platform Shutdown Device  : sifive_test
Firmware Base             : 0x80000000
Firmware Size             : 300 KB
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*,4*,5*,6*,7*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff (I)
Domain0 Region01          : 0x0000000080000000-0x000000008007ffff ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000082200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.10
**Boot HART Base ISA        : rv64imafdc**
Boot HART ISA Extensions  : time
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 54
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

gagachang, 2022-10-20:
@billasurendra 

The supported ISA models of the hart is read from CSR **MISA**.
Maybe the hardware you tested has already supported it ?