riscv-software-src/opensbi #273: qemu+gdb Debug issue

Bigmounten, 2022-10-15:
How to use qemu+gdb to debug opensbi
When i trying to set breakpoint in sbi_init
If i use s
It shows Cannot find bounds of current function
If i use c ,it's just stop there
I'm looking forward to your reply .Thanks.

Reading symbols from /opensbi-1.1/build/platform/generic/firmware/fw_payload.elf...
(gdb) target remote:1234
Remote debugging using :1234
0x0000000000001004 in ?? ()
(gdb) b sbi_init
Breakpoint 1 at 0x800005b4: file /opensbi-1.1/lib/sbi/sbi_init.c, line 467.
(gdb) s
Cannot find bounds of current function
(gdb) c
Continuing.

Dunky-Z, 2022-11-07:
Hi, may be you could use `si` instead of `s` or `step`. `si` works well on my pc.

```bash
Type "apropos word" to search for commands related to "word".
(gdb) file  /path/to/build/platform/generic/firmware/fw_payload.elf
Reading symbols from  /path/to/platform/generic/firmware/fw_payload.elf...
(gdb) target remote :1234
Remote debugging using :1234
0x0000000000001004 in ?? ()
(gdb) b _start
Breakpoint 1 at 0x80000000: file /path/to/opensbi/firmware/fw_base.S, line 50.
(gdb) s
Cannot find bounds of current function
(gdb) si
0x0000000000001008 in ?? ()
(gdb) si
0x000000000000100c in ?? ()
(gdb) step
Cannot find bounds of current function
(gdb) si
0x0000000000001010 in ?? ()
(gdb) si
0x0000000000001014 in ?? ()
(gdb) si
0x0000000000001018 in ?? ()
(gdb) si

Breakpoint 1, _start () at /path/to/firmware/fw_base.S:50
warning: Source file is more recent than executable.
50              MOV_3R  s0, a0, s1, a1, s2, a2
(gdb) si
0x0000000080000004      50              MOV_3R  s0, a0, s1, a1, s2, a2
(gdb) si
0x0000000080000008      50              MOV_3R  s0, a0, s1, a1, s2, a2
(gdb) si
52              call    fw_boot_hart
(gdb) si
fw_boot_hart () at /path/to/firmware/fw_payload.S:22
22              li      a0, -1
(gdb) si
23              ret
```

lbmeng, 2023-07-22:
When you build the OpenSBI binary, you can pass "DEBUG=1" on the command line which generates debugger-friendly bits.

Bigmounten, 2023-07-23:
> When you build the OpenSBI binary, you can pass "DEBUG=1" on the command line which generates debugger-friendly bits.

Thank u so much!