riscv-software-src/opensbi #274: SCMI protocols for power management

JeremyAngCH, 2022-10-16:
If the RISC-V platform is using SCMI protocols for power management, when we try to upstream the code, the SCMI code should be part of the library private to the specific platform (under '/platform/<platform_name>') or it can be a general library ( under '/lib/utils') ?

avpatel, 2022-10-16:
I can be a general library under lib/utils if it useful to other platforms as well.

The SCMI protocol parts can be under lib/utils whereas SCMI transport can be under platform/<xyz>.

Regards,
Anup

JeremyAngCH, 2022-10-17:
Thanks for the reply. The SCMI protocol library should be general enough to be used by other platforms as well.<br>
Is it OK if the SCMI code is adapted/ported from ATF ?

JeremyAngCH, 2022-10-18:
Thanks for the advise.<br>
The SCMI code from ATF is BSD 3-clause, is that OK ? <br>
I have seen some of the libraries used in OpenSBI are licensed under BSD 3-clause as well:
https://github.com/riscv-software-src/opensbi/tree/bd355213bfbb209c047e8cc0df56936f6705477f/lib/utils/libquad

avpatel, 2022-10-21:
Yes, BSD 3-clause should be fine.

atishp04, 2022-10-21:
Here are my 2 cents:
Just skimming through SCMI spec, It has numerous references to PSCI system reset & ACPI power states. Those will be different in RISC-V world. 

If you plan to use SCMI, it can't be standardized in RISC-V in the future due to legal reasons. We had few rounds of discussion around this in 2020 during platform specification. The implementation can be added in OpenSBI but it will be partial implementation at best.

It would be great to have similar messaging protocol for RISC-V. In the beginning, it can be kept in OpenSBI as well. If ther eis more interest from other RISC-V vendors, we can try to standardize through RVI as well.

JeremyAngCH, 2022-10-22:
Thanks. I totally agree with you. The minimal or base SCMI implementation (sufficient for common use cases) will be part of the "3rd party" protocol in OpenSBI. Eventually, it shall be replaced with similar messaging protocol defined and standardized for RISC-V world.

avpatel, 2022-10-22:
We (Ventana) are also interested in a SCMI-like messaging protocol which is RISC-V focused.

Maybe we can call it RISC-V SCMP ?

I would suggest having an initial draft of RISC-V SCMP spec under <opensbi_source>/docs directory. If more RISC-V members are interested then we can RISC-V International can standardise it in future.

Regards,
Anup

JeremyAngCH, 2022-10-25:
SCMP (System Control and Management Protocol) sounds good to me. 
Does it means we are going to have an equivalent "SCP-Firmware" (link below) as well in the future, right ?

https://github.com/ARM-software/SCP-firmware

avpatel, 2022-10-25:
I think having an open RISC-V SCMP specification is a good starting point.

Typically SCP-style power management firmware have a lot of implementation specific details so SoC vendors (just like ARM world) may want to keep it closed source. In fact, the ARM SCP-firmware is mainly used for ARM evaluation boards and other companies (Qualcomm for example) have their own SCP-style firmware.

Regards,
Anup

GuEe-GUI, 2025-02-20:
It just a protocol guy! I have created a issue for PSCI https://github.com/riscv-software-src/opensbi/issues/275, But upstream doesn't seem interested. 
Well, the `Spacemit` RISCV SoC is try to use SCMI for kernel DVFS and PSCI in OpenSBI.