riscv-software-src/opensbi #281: BUG: OpenSBI crashes when passing custom DTB on Qemu

maroueneboubakri, 2022-12-17:
Hello,

Providing a custom DTB through qemu's -dtb option makes OpenSBI randomly crashes on boot.

```
OpenSBI v1.1
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

sbi_trap_error: hart0: trap handler failed (error -2)
sbi_trap_error: hart0: mcause=0x0000000000000005 mtval=0x0000000000000004
sbi_trap_error: hart0: mepc=0x00000000800118ee mstatus=0x8000000a00007800
sbi_trap_error: hart0: ra=0x000000008000393a sp=0x000000008004cea0
sbi_trap_error: hart0: gp=0x0000000000000000 tp=0x000000008004d000
sbi_trap_error: hart0: s0=0x000000008004cf10 s1=0x0000000000000000
sbi_trap_error: hart0: a0=0x0000000000000000 a1=0x000000000000000a
sbi_trap_error: hart0: a2=0x0000000000000001 a3=0x0000000000000001
sbi_trap_error: hart0: a4=0x0000000000000000 a5=0x0000000000000003
sbi_trap_error: hart0: a6=0x0000000000000001 a7=0x0000000000000005
sbi_trap_error: hart0: s2=0x0000000000000000 s3=0x0000000000000000
sbi_trap_error: hart0: s4=0x0000000000000000 s5=0x000000008001d000
sbi_trap_error: hart0: s6=0x0000000000000000 s7=0x0000000000000004
sbi_trap_error: hart0: s8=0x0000000000002000 s9=0x000000008003d6a8
sbi_trap_error: hart0: s10=0x0000000000000000 s11=0x0000000000000000
sbi_trap_error: hart0: t0=0x00000000800076b0 t1=0x0000000000000002
sbi_trap_error: hart0: t2=0x0000000000001000 t3=0x0000000000000010
sbi_trap_error: hart0: t4=0x0000000080016945 t5=0x0000000000000004
sbi_trap_error: hart0: t6=0x0000000000000000
qemu-system-riscv64: terminating on signal 2

```
I didn't spend much time on debugging the issue but it seems that ``fdt_get_address()`` returns 0 sometimes.

Should mention that this is sporadic, should retry many times to successfully boot.

Thanks
Maro

maroueneboubakri, 2022-12-26:
Can always be reproduced as following:

```
$ qemu-system-riscv64 -s -S -m 256M -machine virt -bios $OPENSBI_PATH -dtb  $DTB_PATH -device -serial stdio --kernel $KERNAL_PATH -append "root=/dev/vda rw console=ttyS0" -drive file=$ROOTFS_PATH,format=raw,id=hd0 -device virtio-blk-device,drive=hd0 -smp 2
```

Attach GDB

```
(gdb) target remote 127.0.0.1:1234
Remote debugging using 127.0.0.1:1234
0x0000000000001000 in ?? ()
(gdb) c
Continuing.
```

Result:
```
OpenSBI v1.1
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

sbi_trap_error: hart0: trap handler failed (error -2)
sbi_trap_error: hart0: mcause=0x0000000000000005 mtval=0x0000000000000004
sbi_trap_error: hart0: mtval2=0x0000000000000000 mtinst=0x0000000000000000
sbi_trap_error: hart0: mepc=0x00000000800118ee mstatus=0x8000000a00007800
sbi_trap_error: hart0: ra=0x000000008000393a sp=0x000000008004cea0
sbi_trap_error: hart0: gp=0x0000000000000000 tp=0x000000008004d000
sbi_trap_error: hart0: s0=0x000000008004cf10 s1=0x0000000000000000
sbi_trap_error: hart0: a0=0x0000000000000000 a1=0x000000000000000a
sbi_trap_error: hart0: a2=0x0000000000000001 a3=0x0000000000000001
sbi_trap_error: hart0: a4=0x0000000000000000 a5=0x0000000000000003
sbi_trap_error: hart0: a6=0x0000000000000001 a7=0x0000000000000005
sbi_trap_error: hart0: s2=0x0000000000000000 s3=0x0000000000000000
sbi_trap_error: hart0: s4=0x0000000000000000 s5=0x000000008001d000
sbi_trap_error: hart0: s6=0x0000000000000000 s7=0x0000000000000004
sbi_trap_error: hart0: s8=0x0000000000002000 s9=0x000000008003d6a8
sbi_trap_error: hart0: s10=0x0000000000000000 s11=0x0000000000000000
sbi_trap_error: hart0: t0=0x00000000800076b0 t1=0x0000000000000002
sbi_trap_error: hart0: t2=0x0000000000001000 t3=0x0000000000000010
sbi_trap_error: hart0: t4=0x0000000080016945 t5=0x0000000000000004
sbi_trap_error: hart0: t6=0x0000000000000000
```

Thanks

lbmeng, 2022-12-27:
Please provide your dtb file.

pathakraul, 2022-12-27:
I frequently use custom dtb with qemu and it works fine. These errors are only possible when DTB has some changes which are not compatible with opensbi.

maroueneboubakri, 2022-12-27:
Here is the DTS:

```
/dts-v1/;

/ {
        #address-cells = <0x02>;
        #size-cells = <0x02>;
        compatible = "riscv-virtio";
        model = "riscv-virtio,qemu";

        /*fw-cfg@10100000 {
                dma-coherent;
                reg = <0x00 0x10100000 0x00 0x18>;
                compatible = "qemu,fw-cfg-mmio";
        };*/

        flash@20000000 {
                bank-width = <0x04>;
                reg = <0x00 0x20000000 0x00 0x2000000 0x00 0x22000000 0x00 0x2000000>;
                compatible = "cfi-flash";
        };

        chosen {
                bootargs = "root=/dev/vda rw console=ttyS0 earlycon";
                stdout-path = "/soc/uart@10000000";
				
				opensbi-domains {
					compatible = "opensbi,domain,config";

					mumem: mumem {
						compatible = "opensbi,domain,memregion";
						base = <0x0 0x50000000>;
						order = <10>;
					};

					tmem: tmem {
						compatible = "opensbi,domain,memregion";
						base = <0x0 0x80100000>;
						order = <20>;
					};

					tuart: tuart {
						compatible = "opensbi,domain,memregion";
						base = <0x0 0x10000000>;
						order = <12>;
						mmio;
						devices = <&uart0>;
					};

					allmem: allmem {
						compatible = "opensbi,domain,memregion";
						base = <0x0 0x0>;
						order = <64>;
					};

					tdomain: trusted-domain {
						compatible = "opensbi,domain,instance";
						possible-harts = <&cpu0>;
						/* dev: access to all for now */
						regions = <&tmem 0x7>, <&tuart 0x7>, <&allmem 0x7>, <&mumem 0x7>;
						boot-hart = <&cpu0>;
						next-arg1 = <0x0 0x0>;
						next-addr = <0x0 0x80100000>;
						next-mode = <0x1>;
						system-reset-allowed;
					};

					udomain: untrusted-domain {
						compatible = "opensbi,domain,instance";
[... 226 lines cut ...]
                        riscv,ndev = <0x35>;
                        reg = <0x00 0xc000000 0x00 0x210000>;
                        interrupts-extended = <0x08 0x0b 0x08 0x09 0x06 0x0b 0x06 0x09 0x04 0x0b 0x04 0x09 0x02 0x0b 0x02 0x09>;
                        interrupt-controller;
                        compatible = "sifive,plic-1.0.0\0riscv,plic0";
                        #interrupt-cells = <0x01>;
                        #address-cells = <0x00>;
                };

                clint@2000000 {
                        interrupts-extended = <0x08 0x03 0x08 0x07 0x06 0x03 0x06 0x07 0x04 0x03 0x04 0x07 0x02 0x03 0x02 0x07>;
                        reg = <0x00 0x2000000 0x00 0x10000>;
                        compatible = "sifive,clint0\0riscv,clint0";
                };
        };

		firmware {
			optee {
				compatible = "riscv,optee-td";
				method = "rpmsg";
			};
		};
	
};
```

pathakraul, 2022-12-27:
When you provide the custom dtb, are you providing the `-smp 4`  in the qemu option? Can you try if you didn't

maroueneboubakri, 2022-12-27:
No whatever smp you provide it crashes, I know smp should match the cpus indicated in the DTB as mention in the virt machine description I update the description above.

maroueneboubakri, 2022-12-27:
When issue happens: 

**riscv_cpu_do_interrupt: hart:0, async:0, cause:0000000000000005, epc:0x0000000080012df4, tval:0x0000000000000004, desc=fault_load**

```
riscv_cpu_do_interrupt: hart:0, async:0, cause:0000000000000002, epc:0x0000000080008be2, tval:0x0000000000000000, desc=illegal_instruction
riscv_cpu_do_interrupt: hart:0, async:0, cause:0000000000000002, epc:0x000000008000853e, tval:0x0000000000000000, desc=illegal_instruction
riscv_cpu_do_interrupt: hart:0, async:0, cause:0000000000000002, epc:0x0000000080008566, tval:0x0000000000000000, desc=illegal_instruction
riscv_cpu_do_interrupt: hart:0, async:0, cause:0000000000000002, epc:0x00000000800085b2, tval:0x0000000000000000, desc=illegal_instruction

OpenSBI v1.2
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

riscv_cpu_do_interrupt: hart:0, async:0, cause:0000000000000005, epc:0x0000000080012df4, tval:0x0000000000000004, desc=fault_load
sbi_trap_error: hart0: trap handler failed (error -2)
sbi_trap_error: hart0: mcause=0x0000000000000005 mtval=0x0000000000000004
sbi_trap_error: hart0: mepc=0x0000000080012df4 mstatus=0x8000000a00007800
sbi_trap_error: hart0: ra=0x0000000080003de8 sp=0x0000000080039ea0
sbi_trap_error: hart0: gp=0x0000000000000000 tp=0x000000008003a000
sbi_trap_error: hart0: s0=0x0000000080039f10 s1=0x0000000000000000
sbi_trap_error: hart0: a0=0x0000000000000000 a1=0xffffffffffffffff
sbi_trap_error: hart0: a2=0x000000008001af70 a3=0x0000000080039eb8
sbi_trap_error: hart0: a4=0x0000000000000008 a5=0x0000000000000000
sbi_trap_error: hart0: a6=0x0000000000000000 a7=0x0000000000000044
sbi_trap_error: hart0: s2=0x000000008002aa98 s3=0x0000000000000000
sbi_trap_error: hart0: s4=0x000000008001a1a0 s5=0x0000000000000004
sbi_trap_error: hart0: s6=0x0000000000000000 s7=0x0000000000000004
sbi_trap_error: hart0: s8=0x0000000000002000 s9=0x000000008002aab0
sbi_trap_error: hart0: s10=0x0000000000000000 s11=0x0000000000000000
sbi_trap_error: hart0: t0=0x000000008000830c t1=0x0000000000000011
sbi_trap_error: hart0: t2=0x0000000000001000 t3=0x0000000000000068
sbi_trap_error: hart0: t4=0x000000000000003f t5=0x0000000000000004
sbi_trap_error: hart0: t6=0x0000000000000000
```

lbmeng, 2022-12-30:
Please try the following patch:
https://patchwork.ozlabs.org/project/opensbi/patch/20221230042611.1966361-1-bmeng@tinylab.org/

maroueneboubakri, 2022-12-30:
> Please try the following patch: https://patchwork.ozlabs.org/project/opensbi/patch/20221230042611.1966361-1-bmeng@tinylab.org/

Hi @lbmeng, thanks for the patch, it works, I tested it with qemu 6.2.0 and qemu 7.2.50. I close the issue once the patch gets merged into upstream.

maroueneboubakri, 2022-12-30:
Hi @lbmeng based on the DTS I provided, I noticed that sometimes Linux does not boot.

pathakraul, 2022-12-30:
My observation from start is that whenever coldboot_hart is from trusted domain it was failing, earlier it traps because of fdt address. Now it fails because of illegal instruction .  It always boots when coldboot_hart is from untrusted domain

closed 2023-02-27 by avpatel