riscv-software-src/opensbi #282: RISCV TEE support

liushiwei007, 2022-12-22:
We have developed a security system based on RISCV, in which a part of opensbi code, I wonder whether it can be incorporated.

atishp04, 2022-12-22:
Is it a platform specific solution or generic one ? Can you please provide more details ?

liushiwei007, 2022-12-22:
It's the general part right now，We used hardware technology similar to arm trustzone and a software framework for the optee os  system。We are done with the general purpose part, and the platform-related part is still in development，In this case, opensbi needs to start linux after optee os starts successfully. In addition, two contexts need to be maintained when entering opensbi, one for optee os and one for linux。

liushiwei007, 2022-12-27:
@atishp04   Is this what you need?

liushiwei007, 2022-12-29:
@maroueneboubakri   Have you submitted the relevant code？

liushiwei007, 2022-12-29:
> @maroueneboubakri Have you submitted the relevant code？

I've only recently started contributing open source code,I think I misunderstood the meaning of "mentioned this issue"。I mistakenly thought it was a message from you,，I looked at #5714, which should have been added automatically by you using the here tag.

liushiwei007, 2023-01-07:
@atishp04    Hi,  are there any new updates?

maroueneboubakri, 2023-01-07:
Hi @liushiwei007,
I suggest to join the OpenSBI mailing list and send your request there. I don't think that the repo is well monitored.
Maro

avpatel, 2023-01-07:
I suggest sharing a draft proposal for SBI OP-TEE extension on the OpenSBI and PRS TG mailing list will be a good starting point.

Regards,
Anup

liushiwei007, 2023-01-09:
@maroueneboubakri  @avpatel Ok, thanks。

liushiwei007, 2023-01-09:
> @maroueneboubakri @avpatel Ok, thanks。

hi, I tried to send a message to the mailing list, but the email was returned, is there something wrong with me. The following is the error message:
Original e-mail message : 	　 
Time :	2023-01-09 10:19:30
Subject :	RISC-V TEE submitted
To :	[opensbi@lists.infradead.org](mailto:opensbi@lists.infradead.org)

Bounce reason : 	 
Connection to the remote recipient's server was denied for unknown reason.
SMTP through SDN 37, SMTP: (Proxy)Host lists.infradead.org said 550-Mailing lists do not accept HTML mail. See
550 http://david.woodhou.se/email.html

avpatel, 2023-01-09:
@liushiwei007 please send plain text message on OpenSBI mailing list

liushiwei007, 2023-01-09:
@avpatel Yes, I did send a plain text message

atishp04, 2023-01-12:
@liushiwei007 : Sorry for the delay in reply.

There are two aspects to this. One is the specification and other is the OpenSBI implementation.

1. I believe you sent the patches for the OpenSBI implementation. As the error message from the mailing list says, the format of the email was HTML. The open source mailing lists only accept plain text formatted message. Please try again with text format.

2. Regarding specification, I am assuming you need to introduce new SBI calls. In order to get the standardized within RVI so that entire community can benefit from this, please send the the specification details to tech-prs Task group mailing list (tech-prs@lists.riscv.org). 

If you need some early feedback from the RVI task group, you can present your work in the tech-prs task group meeting as well. Let me know and I can schedule something in upcoming meetings.

liushiwei007, 2023-01-12:
@atishp04 
I think I was misunderstood the “plain text”. I have successfully sent the email. Thank you for your explanation.  I will try to do something.

z3cko, 2023-06-25:
I am just wondering (since i am super interested in this topic myself) - has something ever been sent to the mailing list? what is the status of TEE with openSBI? thanks!

liushiwei007, 2023-06-27:
You can check the record here(https://github.com/riscv-non-isa/riscv-sbi-doc/pull/106). I will keep trying, but the progress may be slow.

gagachang, 2023-07-04:
> I am just wondering (since i am super interested in this topic myself) - has something ever been sent to the mailing list? what is the status of TEE with openSBI? thanks!

Hi @z3cko 

There is another WIP specification called RISC-V **CoVE**: https://github.com/riscv-non-isa/riscv-ap-tee
Several SBI functions are well-defined there.

However, they are focusing on TEE with **hypervisor** extension supported.
We are lack of specification which is without **hypervisor** extension needed.