riscv-software-src/opensbi #283: Rethink PMPs assignment

maroueneboubakri, 2022-12-28:
Hi,

My use case is simple, I'm configuring 2 domains, called trusted and untrusted. The trusted memory region starts at 0x82000000 while the untrusted region starts at 0x80200000. The regions does not overlap.
For this use case few PMP entries are needed (considering PMP to protect OpenSBI etc).

This should work fine in platform with 16 PMP entries, however this is not the case. I noticed that running the SW in trusted domain triggers an Illegal Instruction, by debugging it turned out that, OpenSBI writes to non-existing PMP register. Its address is 0x3c0. This is pmpaddr16. However my platform has only 16 PMP entries.

By looking at the code https://github.com/riscv-software-src/opensbi/blob/master/lib/sbi/sbi_hart.c#L315-L318 The PMP is selected by shifting the desired address.

I guess this should be re-considered. 

Thanks
Maro

maroueneboubakri, 2022-12-28:
@avpatel FYI.

gagachang, 2022-12-29:
This loop should be break since it checks the supported PMP entries of this hart and the index of PMP entry you are configuring.
**pmp_count** should be 16 in your case.
https://github.com/riscv-software-src/opensbi/blob/master/lib/sbi/sbi_hart.c#L302

The PMP address CSR only depends on this index.
https://github.com/riscv-software-src/opensbi/blob/master/lib/sbi/riscv_asm.c#L270

maroueneboubakri, 2022-12-29:
@hschauhan yes that's qemu virt machine and RV64 as target. That's the expectation. 
@gagachang By looking at the code yes.  Need to investigate further what makes the FW writes to PMP < PMP15.

```
	tmem: tmem {
		compatible = "opensbi,domain,memregion";
		base = <0x0 0x82000000>;
		/* 32M */
		order = <25>;
	};
```
Disassembly :
```
00000000800083be <sbi_hart_init>:
...
80008be2:   3c002873                csrr    a6,0x3c0 <-- illegal_instruction
...
```

Replacing **0x82000000** with **0x80100000** works fine.

maroueneboubakri, 2022-12-29:
> Maybe you can check the return value of `sbi_hart_pmp_count()` first.

PMP count is 16.

maroueneboubakri, 2022-12-29:
https://github.com/riscv-software-src/opensbi/blob/master/lib/sbi/riscv_asm.c#L307 is checking against `PMP_COUNT ` which is 64.

lbmeng, 2022-12-29:
Can you replace PMP_COUNT with sbi_hart_pmp_count() in pmp_set() and pmp_get(), and report if this fixed your issue?

gagachang, 2022-12-29:
> 
> Disassembly :
> 
> ```
> 00000000800083be <sbi_hart_init>:
> ...
> 80008be2:   3c002873                csrr    a6,0x3c0 <-- illegal_instruction
> ...
> ```
> 
> Replacing **0x82000000** with **0x80100000** works fine.

Seems the exception happens in ``hart_detect_features()`` ?
If yes, this function uses a skill which temporarily substitutes the trap vector to ``__sbi_expected_trap()``, and reads PMP CSRs to detect valid PMP entries of your platform.

maroueneboubakri, 2022-12-29:
Yes in addition to that, it traps on the following:

```
80008472:   32002573                csrr    a0,mucounteren
8000849a:   30a02573                csrr    a0,0x30a
800084e6:   fb002573                csrr    a0,0xfb0
```
I don't understand how address **0x82000000** is related to all this.

hschauhan, 2022-12-29:
By default, Qemu has 16 PMPs. It seems like the total regions from 2 domains is overflowing 16 entries. It would be interesting to print the number of regions and their details. @maroueneboubakri Can you disable the call to pmp_set() and just print different regions?

hschauhan, 2022-12-29:
> Yes in addition to that, it traps on the following:
> 
> ```
> 80008472:   32002573                csrr    a0,mucounteren
> 8000849a:   30a02573                csrr    a0,0x30a
> 800084e6:   fb002573                csrr    a0,0xfb0
> ```
> 
> I don't understand how address **0x82000000** is related to all this.

Depends on the number of regions being carved out. My guess is that these two domains are carving out regions more than 16. So, if we can dump the regions that would be better. Could you provide your DTS?

hschauhan, 2022-12-29:
@maroueneboubakri I have posted a patch that takes care of checking the pmp indexes before programming. It prints the last index programmed and the index that failed along with the region details. Please see if that works for you.

maroueneboubakri, 2022-12-29:
> @maroueneboubakri I have posted a patch that takes care of checking the pmp indexes before programming. It prints the last index programmed and the index that failed along with the region details. Please see if that works for you.

@hschauhan thank you ! I'll give it a try.

closed 2023-05-12 by maroueneboubakri