riscv-software-src/opensbi #284: QEMU hang with default, test payload

sammy17, 2023-01-20:
I followed the instructions given in the readme to build the generic opensbi and tried to boot it with the following QEMU command (same as the one given in the readme):

```
make CROSS_COMPILE=riscv64-unknown-linux-gnu- PLATFORM_RISCV_XLEN=64 PLATFORM=generic
qemu-system-riscv64 -M virt -m 256M -nographic -bios opensbi/build/platform/generic/firmware/fw_payload.bin
```

**I see that OpenSBI starts but no payload gets executed after that. Is this expected? Is there a test payload that gets executed after OpenSBI in this case?**

Following is what is displayed after I execute the QEMU command and nothing is printed in the terminal after this:
```

OpenSBI v1.2-19-gda5594b
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : riscv-virtio,qemu
Platform Features         : medeleg
Platform HART Count       : 1
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 10000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : sifive_test
Platform Shutdown Device  : sifive_test
Firmware Base             : 0x80000000
Firmware Size             : 212 KB
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x0000000080000000-0x000000008003ffff M: (R,W,X) S/U: ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff M: (R,W,X) S/U: (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000082200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.10
Boot HART Base ISA        : rv64imafdc
Boot HART ISA Extensions  : time
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 54
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

Test payload running

```

lbmeng, 2023-01-20:
As you can see:

`Test payload running`

This indicates that test payload is running.

sammy17, 2023-01-20:
I see, I thought it hangs as it was not exiting QEMU. 
I am trying to boot Linux with opensbi and QEMU, and it was hanging after printing opensbi output (no kernel bootup is shown).
Thanks for clarifying, I will close this issue and file another one if necessary for the Linux bootup issue.

closed 2023-01-20 by sammy17