riscv-software-src/opensbi #289: DOC: Example need to be update in OpenSBI Domain Support

Zweisamkeiten, 2023-02-24:
https://github.com/riscv-software-src/opensbi/commit/001106d19b21cd6443ae7f7f6d4d048d80e9ecac

> Any region of a domain defined in DT node cannot have only M-bits set in access permissions i.e. it cannot be an m-mode only accessible region.

>                 regions = <&tmem 0x0>, <&tuart 0x0>, <&allmem 0x7>;

```c
	if (!((region_access & SBI_DOMAIN_MEMREGION_SU_ACCESS_MASK)
	     & SBI_DOMAIN_MEMREGION_SU_RWX)
	    && ((region_access & SBI_DOMAIN_MEMREGION_M_ACCESS_MASK)
		& SBI_DOMAIN_MEMREGION_M_RWX))
		return SBI_EINVAL;
```

lbmeng, 2023-02-24:
What a timing :)

I've already sent a patch to fix this earlier today.
See [here](https://patchwork.ozlabs.org/project/opensbi/patch/20230224042827.1768931-3-bmeng@tinylab.org/)

Zweisamkeiten, 2023-02-25:
> 

Thank you for your reply. I will close this issue.

closed 2023-02-25 by Zweisamkeiten