riscv-software-src/opensbi #299: initial stack for payload

jiahzhang, 2023-04-02:
Is there ever an initial stack setup/mapped for the kernel? I am seeing an error where because STVEC is not setup yet and linux kernel head.S attempts to execute a c.ldsp we end up looping (stuck) back to start.

jiahzhang, 2023-04-02:
FWIW, this seems to only happen when the payload is fairly large. Maybe opensbi is relying on the initial stack to sit within a page?

avpatel, 2023-04-02:
We are not clearing SP when jumping to the next booting stage so if the kernel tries to access stack pointers without setting up then it will get an access fault.

In other words, there is no harm in GPRs having M-mode addresses when entering S-mode because S-mode can't access M-mode memory.

jiahzhang, 2023-04-02:
Thanks for replying. This seems like some transient GCC bug causing S-mode linux to attempt to access the stack.

closed 2023-04-02 by jiahzhang