riscv-software-src/opensbi #300: Smp Linux hardware support for dual core

ErShubhamYY, 2023-04-06:
We have designed single core and planning to extend it to dual core and boot smp Linux(both core are identical). We have found out that we need following things to be implemented correctly.
1. Cache coherence
2. Atomic instruction to be implemented for dual core
3. Inter processor interrupts(use aclint for that)
4. Memory barrier synchronisation instruction (fence, fence.i)
Apart from that do anything else need to be considered to make dual core smp Linux compatible. We are building smp Linux using buildroot tools.

avpatel, 2023-04-06:
If you have mix of cache-coherent and non-cache-coherent devices on your platform then you also need Svpbmt and Zicbom ISA extensions.

ErShubhamYY, 2023-04-07:
> If you have mix of cache-coherent and non-cache-coherent devices on your platform then you also need Svpbmt and Zicbom ISA extensions.

Do we have to implement svpbmt and zicbom even if we hardcode pbmte , cbie and cbcfe bit to 0 in menvcfg csr register?