riscv-software-src/opensbi #310: How to enter M-mode?

SamHSmith, 2023-06-07:
I have a device that boots using opensbi. But I actually don't want to use opensbi. How do I enter M mode with my own code once the machine is booted so that I can take over the machine? What SBI call can promote me to machine mode. Please tell me you thought of this.

z3cko, 2023-06-25:
Did you actually read what OpenSBI is about? (see the [README](https://github.com/riscv-software-src/opensbi/blob/master/README.md)) - "A platform-specific firmware running in M-mode and a bootloader, a hypervisor or a general-purpose OS executing in S-mode or HS-mode."

SamHSmith, 2023-06-27:
Yea I got it. I'm going to do my best to eliminate it from my platform.

closed 2023-06-27 by SamHSmith

rejsys, 2023-08-21:
There is a hack for that specific scenario mentioned in #317. Even if OpenSBI is not mend to do this, its sometimes needed to if you want your own application to start in m-mode.