riscv-software-src/opensbi #316: Please push branch containing v1.3.1

vagrantc, 2023-07-21:
There does not appear to be a branch that contains v1.3.1, which makes it rather complicated to actually fetch the tag.

  https://github.com/riscv-software-src/opensbi/tree/v1.3.1

  This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository.

Please push a 1.3.x branch or some such with the tag!

lbmeng, 2023-07-22:
Please do "git fetch --tags"

vagrantc, 2023-07-22:
While "git fetch --tags" was not able to pull the tag in, but "git fetch --tags origin" did work for me. But that is moot now, as
someone pushed a "release-1.3.1" branch containing the tag, so having to go out of the way to run specific commands to pull all tags is no longer necessary.

closed 2023-07-22 by vagrantc