riscv-software-src/opensbi #320: envcfg only present in privileged 1_12, OpenSBI detects its presence but Linux kvm does not

mingyuan-xia, 2023-08-09:
[mhs]envcfg are present from privileged spec 1.12. OpenSBI [explicitly detects](https://github.com/riscv-software-src/opensbi/blob/94197a8c49751eabe73ff2849fce183ac8ead3fd/lib/sbi/sbi_hart.c#L836) that to determine the privileged level of harts. But [Linux KVM would use it anyway](https://elixir.bootlin.com/linux/v6.1.44/source/arch/riscv/kvm/vcpu.c#L820). Should we also detect its presence before using it @avpatel ?

avpatel, 2023-08-09:
KVM uses explicitly because both H-extension and envcfg CSRs were introduced in Priv v1.12 so if H-extension is present then envcfg CSRs are also present.

OpenSBI has no dependency on H-extension or particular Priv spec version so OpenSBI has to detect envcfg CSRs.

mingyuan-xia, 2023-08-09:
Thanks for the clarification.

closed 2023-08-09 by mingyuan-xia