riscv-software-src/opensbi #324: Detect extensions from $menvcfg

yong-xuan, 2023-09-01:
Hi @avpatel,

Currently OpenSBI can only detect extensions by test and set their CSR. Maybe we can detect more extensions from $menvcfg?
```c
static int hart_detect_features(struct sbi_scratch *scratch)
{
    ...
    if (hfeatures->priv_version >= SBI_HART_PRIV_VER_1_12) {
        csr_write(CSR_MENVCFG, ~0);
        csr_read(CSR_MENVCFG, menvcfg_val);

        if (menvcfg_val | ENVCFG_CBZE)
            __sbi_hart_update_extension(hfeatures,
					SBI_HART_EXT_ZICBOZ, true);
        ...
    }
    ...
}
```

atishp04, 2023-09-29:
Generally, we should move away from WARL based detection mechanism as it will be unmanageable in the future with more number of extensions.

yong-xuan, 2023-10-03:
Hi @atishp04,
Thank you!
OpenSBI can detect extensions from the ISA string since https://github.com/riscv-software-src/opensbi/commit/d72f5f17478dfef3df817339a0f02cfdb77a1dc1. It appears that we can consider removing all extension detection within `hart_detect_features()`?