riscv-software-src/opensbi #328: Enable performance counters

xydas97, 2023-09-28:
Hello, I would need some help enabling performance counters for rocket core of rocketchip. I am currently trying to directly use csr_write() on MCOUNTEREN, SCOUNTEREN, MCOUNTINHIBIT and MHPMEVENT in the sbi_init.c file. However I see that the values are not set in those counters. Are there any functions that I could use or any direction on how to monitor TLB misses with performance counters?

lyctw, 2023-12-20:
Hi xydas97,

You'll need a [PMU node](https://github.com/riscv-software-src/opensbi/blob/master/docs/pmu_support.md#sbi-pmu-device-tree-bindings) that provides event id to selector value (the value written to MHPMEVENT) mapping,  then you can use perf-tool in Linux:

```
CONFIG_RISCV_PMU=y
CONFIG_RISCV_PMU_SBI=y
```
See drivers/perf/riscv_pmu_sbi.c for details.

PMU SBI interfaces are documented in the [SBI spec.](https://github.com/riscv-non-isa/riscv-sbi-doc/blob/master/src/ext-pmu.adoc)

atishp04, 2023-12-20:
I think it's duplicate of 
https://github.com/riscv-software-src/opensbi/issues/249

Closing.

closed 2023-12-20 by atishp04