riscv-software-src/opensbi #335: Too small fw_heap_size makes OpenSBI corrupt memory

eenurkka, 2023-11-27:
Heap minimum appears to be 16k. This is not documented anywhere nor checked anywhere.

Otherwise sbi_heap.c fails:

hpctrl.hksize = hpctrl.size / HEAP_HOUSEKEEPING_FACTOR;
hpctrl.hksize &= ~((unsigned long)HEAP_BASE_ALIGN - 1);

8k:  (0x2000 / 16) & ~(1024 - 1) = 0      (Fail!)
16k: (0x4000 / 16) & ~(1024 - 1) = 0x400  (Ok)

For example, with fw_heap_size = 8k, hpctrl.hksize gets to be zero making the OpenSBI corrupt memory wildly. Everything goes bust and boom in that case.

mateoconlechuga, 2023-12-08:
Personally I feel that the firmware linker script should have sections for the stack, heap, and scratch whose sizes are controlled through menuconfig for the platform. What are the thoughts on this?

avpatel, 2023-12-09:
All three per-CPU stack, per-CPU scratch, and system wide heap are scaled at boot time based on the number of CPUs. This allows us to use the same FW binary on multiple platforms. We can further improve this by allowing platforms to override heap size using a DT property but this is not available yet.

mateoconlechuga, 2023-12-09:
- The same firmware binary doesn't work on multiple platforms though since the platform is responsible for allocating the heap size based on the expected number of cores (like the template here just shows a magic "1" value here representing 1 hart: https://github.com/riscv-software-src/opensbi/blob/master/platform/template/platform.c#L155).

The number of cores and the per-hart stack size could be passed to the linker script - right now the developer of the platform has to know the number of cores anyway since it's a platform-specific define at compile time.