riscv-software-src/opensbi #336: TH1520（C910）boot information: CPU1、CPU2、CPU3: failed to come online ;  only CPU0 succeed to come online

ixgbe01, 2023-12-01:
build LicheePi4A images link：https://github.com/armbian/build.git
build command:      **sudo ./compile.sh BOARD=licheepi-4a BRANCH=edge RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_CONFIGURE=yes COMPRESS_OUTPUTIMAGE=sha,gpg,xz**

Linux6.6 kernel link:  https://github.com/revyos/th1520-linux-kernel.git
branch:  th1520-v6.6.3

opensbi link:  https://github.com/riscv-software-src/opensbi.git
branch:  master

how to update opensbi binary? you can refer to the following fix:
thead: update linux to 6.6-rc5, update opensbi binary
https://github.com/armbian/build/commit/3098c2f0cc4441eef052eb825d7f39c1f28a1c99

LicheePi4A TH1520 boot information txt:
[licheePi4A_linux6_6_dmesg_official_opensbi.txt](https://github.com/riscv-software-src/opensbi/files/13521097/licheePi4A_linux6_6_dmesg_official_opensbi.txt)

[    0.090218] smp: Bringing up secondary CPUs ...
[    1.102092] CPU1: failed to come online
[    2.129308] CPU2: failed to come online
[    3.156496] CPU3: failed to come online
[    3.160519] smp: Brought up 1 node, 1 CPU

However if we use https://github.com/revyos/opensbi.git （branch: th1520-v1.3.1） , LicheePi4A TH1520 boot information is shown as following:
[licheePi4A_linux_6_6_dmesg_revyos_opensbi.txt](https://github.com/riscv-software-src/opensbi/files/13521130/licheePi4A_linux_6_6_dmesg_revyos_opensbi.txt)

[    0.090069] smp: Bringing up secondary CPUs ...
[    0.118766] cpu1: Ratio of byte access time to unaligned word access is 6.14, unaligned accesses are fast
[    0.150904] cpu2: Ratio of byte access time to unaligned word access is 6.14, unaligned accesses are fast
[    0.183032] cpu3: Ratio of byte access time to unaligned word access is 6.14, unaligned accesses are fast
[    0.193054] smp: Brought up 1 node, 4 CPUs

pdp7, 2023-12-01:
@RevySR Do you know how to make use of the changes in [zero_stage_boot](https://github.com/T-head-Semi/zero_stage_boot)?  I'm not sure about the steps to deploy it onto a board.

RevySR, 2023-12-01:
> @RevySR Do you know how to make use of the changes in [zero_stage_boot](https://github.com/T-head-Semi/zero_stage_boot)? I'm not sure about the steps to deploy it onto a board.

I don't know. Ren Guo knows how to do it

guoren83, 2023-12-01:
> build LicheePi4A images link：https://github.com/armbian/build.git build command: **sudo ./compile.sh BOARD=licheepi-4a BRANCH=edge RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_CONFIGURE=yes COMPRESS_OUTPUTIMAGE=sha,gpg,xz**
> 
> Linux6.6 kernel link: https://github.com/revyos/th1520-linux-kernel.git branch: th1520-v6.6.3
> 
> opensbi link: https://github.com/riscv-software-src/opensbi.git branch: master
> 
> how to update opensbi binary? you can refer to the following fix: thead: update linux to 6.6-rc5, update opensbi binary [armbian/build@3098c2f](https://github.com/armbian/build/commit/3098c2f0cc4441eef052eb825d7f39c1f28a1c99)
> 
> LicheePi4A TH1520 boot information txt: [licheePi4A_linux6_6_dmesg_official_opensbi.txt](https://github.com/riscv-software-src/opensbi/files/13521097/licheePi4A_linux6_6_dmesg_official_opensbi.txt)
> 
> [ 0.090218] smp: Bringing up secondary CPUs ... [ 1.102092] CPU1: failed to come online [ 2.129308] CPU2: failed to come online [ 3.156496] CPU3: failed to come online [ 3.160519] smp: Brought up 1 node, 1 CPU
> 
> However if we use https://github.com/revyos/opensbi.git （branch: th1520-v1.3.1） , LicheePi4A TH1520 boot information is shown as following: [licheePi4A_linux_6_6_dmesg_revyos_opensbi.txt](https://github.com/riscv-software-src/opensbi/files/13521130/licheePi4A_linux_6_6_dmesg_revyos_opensbi.txt)
> 
> [ 0.090069] smp: Bringing up secondary CPUs ... [ 0.118766] cpu1: Ratio of byte access time to unaligned word access is 6.14, unaligned accesses are fast [ 0.150904] cpu2: Ratio of byte access time to unaligned word access is 6.14, unaligned accesses are fast [ 0.183032] cpu3: Ratio of byte access time to unaligned word access is 6.14, unaligned accesses are fast [ 0.193054] smp: Brought up 1 node, 4 CPUs
We've deleted reset-sample: 
https://github.com/riscv-software-src/opensbi/commit/d1e0f7f25b2f1527425942dfc327f1d4a61bbff4

guoren83, 2023-12-02:
> The problem is that upstream OpenSBI is missing a HSM driver for this SoC. With or without [zero_stage_boot](https://github.com/T-head-Semi/zero_stage_boot), a proper HSM driver is still needed for CPU power management (hotplug, cpuidle). And once OpenSBI has a real HSM driver, you don't need any assistance from zero_stage_boot for SMP bringup.

Yes, the th1520 needs an SoC platform HSM driver. The zero_stage_boot is just for the CPU FPGA prototype to replace the reset_sample.

guoren83, 2023-12-02:
> > @RevySR Do you know how to make use of the changes in [zero_stage_boot](https://github.com/T-head-Semi/zero_stage_boot)? I'm not sure about the steps to deploy it onto a board.
> 
> I don't know. Ren Guo knows how to do it

Before zero_stage_boot get into the first-stage bootloader, you could revert the removing reset-sample patch first.

ixgbe01, 2023-12-04:
> ~armbian kernel config is wrong.~
> 
> The commit remove fdt_reset_thead [d1e0f7f](https://github.com/riscv-software-src/opensbi/commit/d1e0f7f25b2f1527425942dfc327f1d4a61bbff4)
> 
> https://github.com/T-head-Semi/zero_stage_boot

With the help of your advice, CPU1, CPU2, and CPU3 could succeed to come online normally. Thank you so much!