riscv-software-src/opensbi #339: semihosting_write bug in generic

zxtxin, 2023-12-15:
semihosting_write is called in semihosting_puts which expects that semihosting_write returns the length it writes.
Length minus the return value of syswrite trap which is normally the true length the syscall writes according to the convention is returned in the implementation of semihosting_write. So semihosting_write returns zero actually.

When I am debuging the opensbi with a debugger attached, I get infinite loop of printing 'OpenSBI 1.3.1'.

wxjstz, 2023-12-20:
[syscall normally return the number of bytes that are not written](https://github.com/openocd-org/openocd/blob/master/src/target/semihosting_common.c#L1365)
semihosting_write in opensbi normally return the number of bytes that has been written
semihosting_puts in opensbi normally return the number of bytes that has been written

Regards,
Xiang W

zxtxin, 2023-12-21:
In linux, the syscall write returns the true length it writes.
I don't know why there is a difference in the implementation of openocd.

zxtxin, 2023-12-21:
I'm using the cklink from t-head to debug a riscv core C908. Its implementation complys with the convention of linux write

cp0613, 2024-01-10:
You can download the latest XuanTie Debug Server from [https://www.xrvm.cn/](https://www.xrvm.cn/) and add the `-ls` parameter at startup to use semihosting, such as `# DebugServerConsole - prereset - ls`

wxjstz, 2024-01-12:
XuanTie implements a strange SYS_WRITE. Is this weird stuff going to spread to all software?