riscv-software-src/opensbi #340: [help] how to run on non-qemu platform

yf13, 2023-12-21:
Hi,

I downloaded v1.3.1 release and can smoothly build and run the test payload on qemu-system-riscv64, this is great!

However, I am wondering where I can find guides to use it on a real hardware? I do have T-Head c908 based SoC and I am wondering where I can find docs / guides to quickly build and run the test payload on that?

Regards,
yf13

SiFiveHolland, 2023-12-28:
There is no Makefile dependency to cause the firmware to be rebuilt when changing `FW_TEXT_START`. You could `touch firmware/fw_base.ldS` or delete the build directory to force a rebuild.

yf13, 2023-12-30:
@SiFiveHolland thanks a lot. I got correct FW_TEXT_START with clean build now.

closed 2023-12-30 by yf13