riscv-software-src/opensbi #354: Firmware SMP boot bug

ezelioli, 2024-05-10:
The cold-boot-hart lottery in `fw_base.S` has recently been modified by using the __boot_status_ variable instead of the __relocate_lottery_ variable. This introduces a data race since the __boot_status_ variable is also written by the cold-boot hart to indicate the current boot stage without waiting for the other cores to have completed the lottery. If a non-boot hart enters the lottery when the boot-hart has already completed relocation __boot_status_ is overwritten and the non-boot hart is stuck forever in __wait_for_boot_hart_.
The change has been introduced in [this](https://github.com/riscv-software-src/opensbi/commit/8151105af5e4bd5d2a17fe0babd6cff3b36496fb) commit.

avpatel, 2024-05-12:
Can you send a patch to fix this?

avpatel, 2024-05-12:
We have mailing list based patch review so it would be great if you can send these patchs to OpenSBI mailing list.

You need to join OpenSBI mailing list using following link
http://lists.infradead.org/mailman/listinfo/opensbi

Make sure you use "git send-email" to send the patches.

Regards,
Anup

wxjstz, 2024-09-20:
Patch here: https://lists.infradead.org/pipermail/opensbi/2024-September/007375.html

Regards,
Xiang W

wxjstz, 2024-12-11:
Fixes: 8151105af5e4 ("firmware: fw_base.S: Remove _relocate_lottery")