riscv-software-src/opensbi #357: firmware base is not the value of FW_TEXT_START( but 0x80000000)

AllenZF, 2024-06-07:
commit d962db280725b03a0340e05a07e4c85c93f35bc5
In WSL:
opensbi$ make all LLVM=1 PLATFORM=generic PLATFORM_RISCV_XLEN=64 FW_TEXT_START=0
opensbi$ make LLVM=1 run PLATFORM=generic
[image]
then,
opensbi$ make LLVM=1 clean
opensbi$ make all LLVM=1 PLATFORM=generic PLATFORM_RISCV_XLEN=64 FW_TEXT_START=0x1000
opensbi$ make LLVM=1 run PLATFORM=generic
[image]
There is nothing to be printed.

tswaehn, 2024-08-16:
if you want to start at **0x80000000** then you should use `qemu-system-riscv32`

if you really stick to `qemu-sytem-riscv64` then this should be **0x80200000**

closed 2024-09-10 by AllenZF

local-h0st, 2025-03-24:
That's what I encountered..I changed the FW_TEXT_START address but not worked, still 0x80000000, any solutions? thx!

SiFiveHolland, 2025-03-24:
The reason `FW_TEXT_START=0x1000` does not work is because of this code in the linker script, which tries to make a NAPOT region for the firmware text/rodata:
https://github.com/riscv-software-src/opensbi/blob/b2e8e6986d676005f23fc5e3b3845b027fc4330f/firmware/fw_base.ldS#L59-L64
This only works if `FW_TEXT_START` is aligned to at least that same power of 2. For the generic platform the text/rodata size is between 128-256 KiB, so `FW_TEXT_START` must be a multiple of 256 KiB. I just tried the steps from the original report (but with `FW_TEXT_START=0x40000`) and it works fine.

@local-h0st don't forget that you need to do a clean build to change `FW_TEXT_START`.

local-h0st, 2025-03-24:
> The reason `FW_TEXT_START=0x1000` does not work is because of this code in the linker script, which tries to make a NAPOT region for the firmware text/rodata:
> 
> [opensbi/firmware/fw_base.ldS](https://github.com/riscv-software-src/opensbi/blob/b2e8e6986d676005f23fc5e3b3845b027fc4330f/firmware/fw_base.ldS#L59-L64)
> 
> Lines 59 to 64 in [b2e8e69](/riscv-software-src/opensbi/commit/b2e8e6986d676005f23fc5e3b3845b027fc4330f)
> 
>  /* 
>   * PMP regions must be to be power-of-2. RX/RW will have separate 
>   * regions, so ensure that the split is power-of-2. 
>   */ 
>  . = ALIGN(1 << LOG2CEIL((SIZEOF(.rodata) + SIZEOF(.text) 
>  			+ SIZEOF(.dynsym) + SIZEOF(.rela.dyn)))); 
> 
> This only works if `FW_TEXT_START` is aligned to at least that same power of 2. For the generic platform the text/rodata size is between 128-256 KiB, so `FW_TEXT_START` must be a multiple of 256 KiB. I just tried the steps from the original report (but with `FW_TEXT_START=0x40000`) and it works fine.
> [@local-h0st](https://github.com/local-h0st) don't forget that you need to do a clean build to change `FW_TEXT_START`.

Really thanks for your support, I just cleaned whole project and rebuild, It works! Thanks again!