riscv-software-src/opensbi #363: questions about OpenSBI Version 1.5

ji2yoon-jo, 2024-07-10:
I downloaded and ported your latest version of OpenSBI Version 1.5, but during kernel booting, the cause is not stamped on the mcause variable in the sbi_trap_handler () function, but the opensbi address(loaded phisical address) is included, so it cannot be booted.
What's the problem?

ji2yoon-jo, 2024-07-10:
[openSBI version 1.3]
[sbi_trap_handler] mcause=0x9

[openSBI version 1.5]
[sbi_trap_handler] mcause=0x7ff40000

avpatel, 2024-07-10:
Need more information about the issue such as detail bootlog, location of sources, etc.

ji2yoon-jo, 2024-07-10:
OpenSBI v1.5-12-g4a5d11d
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Boot Hart0  Next Address  : 0x0000000080200000
Boot Hart0  Next Arg1     : 0x0000000083000000
Platform Name             : Samsung 
Platform Features         : medeleg
Platform HART Count       : 4
Platform IPI Device       : aia-imsic
Platform Timer Device     : aclint-mtimer @ 1000000Hz
Platform Console Device   : onboot_uart
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x7ff40000
Firmware Size             : 349 KB
Firmware RW Offset        : 0x40000
Firmware RW Size          : 93 KB
Firmware Heap Offset      : 0x4d000
Firmware Heap Size        : 41 KB (total), 2 KB (reserved), 12 KB (used), 26 KB (free)
Firmware Scratch Size     : 4096 B (total), 408 B (used), 3688 B (free)
Runtime SBI Version       : 2.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*
Domain0 Region00          : 0x0000000009000000-0x0000000009003fff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x000000000c004000-0x000000000c007fff M: (I,R,W) S/U: ()
Domain0 Region02          : 0x0000000005000000-0x000000000500ffff M: (I,R,W) S/U: ()
Domain0 Region03          : 0x000000007ff80000-0x000000007ff9ffff M: (R,W) S/U: ()
Domain0 Region04          : 0x000000007ff40000-0x000000007ff7ffff M: (R,X) S/U: ()
Domain0 Region05          : 0x0000000000000000-0xffffffffffffffff M: () S/U: (R,W,X)
Domain0 Next Address      : 0x0000000020200000
Domain0 Next Arg1         : 0x0000000023000000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.12
Boot HART Base ISA        : rv64imafdcvhx
Boot HART ISA Extensions  : smaia,smstateen,sscofpmf,zihpm,zicboz,zicbom,svpbmt,sdtrig,svade,smwg,smwgd
Boot HART PMP Count       : 8
Boot HART PMP Granularity : 12 bits
Boot HART PMP Address Bits: 33
Boot HART MHPM Info       : 8 (0x000007f8)
Boot HART Debug Triggers  : 8 triggers
Boot HART MIDELEG         : 0x0000000000003666
Boot HART MEDELEG         : 0x0000000000f0b508
[sbi_trap.c sbi_trap_handler()] mcause=0x7ff40000 mcause&MCAUSE_IRQ_MASK=0x0

ji2yoon-jo, 2024-07-10:
> OpenSBI v1.5-12-g4a5d11d
>
> / __ \ / __**| _ _ _| | | | |_ __ ___ _ __ | (**_ | |_) || | | | | | '_ \ / _ \ '_ \ ___ | _ < | | | |**| | |_) | __/ | | |____) | |_) || |_ __**/| .**/ _**|_| [...]

ji2yoon-jo, 2024-07-10:
int rc = SBI_ENOTSUPP;
	const char *msg = "trap handler failed";
	struct sbi_scratch *scratch = sbi_scratch_thishart_ptr();
	const struct sbi_trap_info *trap = &tcntx->trap;
	struct sbi_trap_regs *regs = &tcntx->regs;
	ulong mcause = tcntx->trap.cause;   // version 1.5
	//ulong mcause = csr_read(CSR_MCAUSE);  // version 1.3

ji2yoon-jo, 2024-07-10:
ulong mcause = tcntx->trap.cause;   // version 1.5 when patch 1.5 is applied, the mcause value looks strange here.

ji2yoon-jo, 2024-07-10:
> ulong mcause = tcntx->trap.cause; // version 1.5 when patch 1.5 is applied, the mcause value looks strange here.

do I need a kernel patch?

avpatel, 2024-07-10:
We are already saving the `mcause` CSR value in `tcntx->trap.cause`.
(Refer, https://github.com/riscv-software-src/opensbi/blob/master/firmware/fw_base.S#L538)

ji2yoon-jo, 2024-07-10:
What I'm curious about is that if you actually print it out as a log, it contains a different value.
It's strange why it has a different value.

[openSBI version 1.3]
[sbi_trap_handler] mcause=0x9

[openSBI version 1.5]
[sbi_trap_handler] mcause=0x7ff40000

ji2yoon-jo, 2024-07-10:
> What I'm curious about is that if you actually print it out as a log, it contains a different value. It's strange why it has a different value.
> 
> [openSBI version 1.3] [sbi_trap_handler] mcause=0x9
> 
> [openSBI version 1.5] [sbi_trap_handler] mcause=0x7ff40000

if i apply the released patch code(v1.5), this problem is occurring, so please let me know which part to look at.

ji2yoon-jo, 2024-07-10:
I will check carefully and inquire.

closed 2024-07-10 by ji2yoon-jo