riscv-software-src/opensbi #365: opensbi failed to build

ElyesH, 2024-07-14:
I've used c23 dialect to build test coreboot and got the following error:
`    CC         ramstage/arch/riscv/opensbi.o
In file included from 3rdparty/opensbi/include/sbi/fw_dynamic.h:46,
                 from src/arch/riscv/opensbi.c:8:
3rdparty/opensbi/include/sbi/sbi_types.h:47:33: error: two or more data types in declaration specifiers
   47 | typedef int                     bool;
      |                                 ^~~~
3rdparty/opensbi/include/sbi/sbi_types.h:47:1: error: useless type name in empty declaration [-Werror]
   47 | typedef int                     bool;
      | ^~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:423: /cb-build/coreboot-gerrit.0/gcc/EMULATION_QEMU_RISCV_RV64_/ramstage/arch/riscv/opensbi.o] Error 1
make[2]: Leaving directory '/home/coreboot/node-root/workspace/coreboot-gerrit'
`
https://qa.coreboot.org/job/coreboot-gerrit/260415/testReport/junit/(root)/gcc/EMULATION_QEMU_RISCV_RV64_/

Thx

tswaehn, 2024-08-16:
what are the make parameters that make opensbi fail to build?

ElyesH, 2024-09-15:
The problem is that you unconditionally `typedef bool`  in this file `opensbi/include/sbi/sbi_types.h`! 
C23 already have `bool` .

Please use soothing like : 
`#if __STDC_VERSION__ <= 201710L
typedef _Bool   bool;
#endif`

tswaehn, 2025-04-11:
Do you have a suggestion to fix it? Maybe you want to create a merge request.

lu-zero, 2025-04-11:
What was suggested in the comment above mine should work. I'll prepare something this weekend if nobody beats me at it.

lu-zero, 2025-04-12:
d539d64a23bbd64ba324eed595449bcc9c7be0cb fixed it in master. It can be closed.