riscv-software-src/opensbi #366: error: "__always_inline" redefined [-Werror]

ElyesH, 2024-07-15:
We can't update opensbi submodule to upstream (since 2022 !) as it fail to build : 
`    CC         ramstage/arch/riscv/misc.o
    CC         ramstage/arch/riscv/opensbi.o
In file included from 3rdparty/opensbi/include/sbi/fw_dynamic.h:46,
                 from src/arch/riscv/opensbi.c:8:
3rdparty/opensbi/include/sbi/sbi_types.h:72:9: error: "__always_inline" redefined [-Werror]
   72 | #define __always_inline inline __attribute__((always_inline))
      |         ^~~~~~~~~~~~~~~
In file included from <command-line>:
./src/commonlib/bsd/include/commonlib/bsd/compiler.h:58:9: note: this is the location of the previous definition
   58 | #define __always_inline inline __attribute__((__always_inline__))
      |         ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:423: /cb-build/coreboot-gerrit.0/gcc/EMULATION_QEMU_RISCV_RV64_/ramstage/arch/riscv/opensbi.o] Error 1
make[2]: Leaving directory '/home/coreboot/node-root/workspace/coreboot-gerrit'`

(see https://qa.coreboot.org/job/coreboot-gerrit/260455/testReport/junit/(root)/gcc/EMULATION_QEMU_RISCV_RV64_/ )

ElyesH, 2024-09-15:
No body at home?

avpatel, 2024-09-15:
@ElyesH Sorry for the delay. Most of the discussions generally happen on mailing list so Github issues get overlooked.

I see you have raised a GitHub PR as well. I will send your patch to the mailing list.

closed 2024-09-27 by avpatel