riscv-software-src/opensbi #367: Why ld is leaving a hole in memory?

thomas-coding, 2024-08-08:
Hi 

opensbi using a large amount of memory now, and there is a memory hole, is there any way to avoid it？thanks.

opensbi/firmware/fw_base.ldS
```
PROVIDE(_rodata_end = .);

/* End of the read-only data sections */

/*
* PMP regions must be to be power-of-2. RX/RW will have separate
* regions, so ensure that the split is power-of-2.
*/
. = ALIGN(1 << LOG2CEIL((SIZEOF(.rodata) + SIZEOF(.text)
		+ SIZEOF(.dynsym) + SIZEOF(.rela.dyn))));

PROVIDE(_fw_rw_start = .);
```

Sections sample:
rela.dyn end 0x200674e8, data start  0x20080000, there is a **98K** memory hole

```
Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         000185a0  0000000020040000  0000000020040000  00001000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .rodata       000029a8  0000000020059000  0000000020059000  0001a000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .dynsym       00000210  00000000200658f8  00000000200658f8  000268f8  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  3 .dynstr       0000019f  000000002005b9a8  000000002005b9a8  0001c9a8  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  4 .hash         000000a4  000000002005bb48  000000002005bb48  0001cb48  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  5 .gnu.hash     000000b4  000000002005bbf0  000000002005bbf0  0001cbf0  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  6 .eh_frame     00008bd4  000000002005bca8  000000002005bca8  0001cca8  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  7 .eh_frame_hdr 0000107c  000000002006487c  000000002006487c  0002587c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  8 .rela.dyn     000014e8  0000000020066000  0000000020066000  00027000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  9 .data         00001398  0000000020080000  0000000020080000  00029000  2**3
                  CONTENTS, ALLOC, LOAD, DATA
 10 .dynamic      00000110  0000000020081398  0000000020081398  0002a398  2**3
                  CONTENTS, ALLOC, LOAD, DATA
 11 .got          000000a8  00000000200814a8  00000000200814a8  0002a4a8  2**3
                  CONTENTS, ALLOC, LOAD, DATA
 12 .got.plt      00000010  0000000020081550  0000000020081550  0002a550  2**3
                  CONTENTS, ALLOC, LOAD, DATA
 13 .bss          00002c90  0000000020082000  0000000020082000  0002a560  2**3
                  ALLOC
```

thomas-coding, 2024-09-03:
Any suggestions?

SiFiveHolland, 2024-09-03:
This is so we can use PMP to protect the read-only part of the firmware. The easiest way to reduce the firmware size is to disable unused drivers in Kconfig (`make menuconfig`). If you can reduce the firmware RO size by 30 KiB, the RW alignment will jump down from 256 KiB to 128 KiB and fill in the hole.

avpatel, 2024-09-03:
Platform vendors can also keep separate defconfig for such things in their fork of OpenSBI repo.

thomas-coding, 2024-09-04:
Considering future expansion(like adding secure os support, power control, etc), it is not easy to limit the size of the firmware
Anyway, I understand the limitations here, thanks