riscv-software-src/opensbi #369: understanding -PIE

tswaehn, 2024-08-16:
just want to understand why `-PIE` is hard coded.

if a bare metal build should be created, then this option prevents it. 

thus shouldnt the `-PIE` option be optional?

SiFiveHolland, 2024-08-16:
The OpenSBI firmware binary is no less bare metal because it is a PIE. OpenSBI applies relocations to itself as part of its entry assembly code, so there is no external dependency to load the firmware.

What problems do you experience when you don't patch OpenSBI?

tswaehn, 2024-08-17:
> What problems do you experience when you don't patch OpenSBI?

Actually my toolchain does not support PIE. I should figure out how to rebuild with PIE if needed. That would be ok.

I appreciate the self relocation and I am a fan of flexibility. Just wondering if optional use of PIE (depending of toolchain capabilities) would be an option.

tswaehn, 2024-08-30:
Just wondering, if we could make the PIE option optional. So we could avoid forking the repo, as mentioned [here](https://github.com/zephyrproject-rtos/zephyr/issues/68133#issuecomment-2321846799).

mikey, 2024-08-30:
I think you just want to submit a patch to opensbi to make PIE an option, or even detect what the toolchain supports. The compile will need to specify FW_TEXT_START (or you could have a default of 0x80000000, like opensbi used to have)

SiFiveHolland, 2024-08-30:
Were you unable to update your toolchain to support PIE? OpenSBI is not the only firmware to be compiled as a PIE. For example, U-Boot is as well. This seems to be a deficiency in your toolchain.

tswaehn, 2024-08-31:
I am using the standard toolchain if Zephyr OS. Same actually goes for all my other toolchains. Usually I don't need PIE.