riscv-software-src/opensbi #370: Access to time CSR is trap-emulated even when scounteren.TM is clear

LekKit, 2024-08-20:
Seems that OpenSBI is emulating missing CSRs on illegal instruction traps, for example unprivileged `time` counter via access to `aclint-mtimer` device.
This is not always correct since the S-mode may explicitly clear `scounteren` bits to disable unprivileged counters and receive illegal instruction traps.

Currently OpenSBI ignores that, and it breaks my virtualization solution that relies on running the guest in U-mode and trapping on privileged operations (including the timer access). It explicitly clears `scounteren` but OpenSBI does me a disservice and allows the virtualized guest to read host timer which effectively breaks the guest.

It would be also good if anyone has an idea how to tell older OpenSBI versions to give up on instruction emulation and just delegate illegal instruction traps to S-mode since a fixed OpenSBI version will take a long time to get on actual hardware.

[image]

LekKit, 2024-08-21:
This line confirms my thoughts:
https://github.com/riscv-software-src/opensbi/blob/b9c091ed890224dc83a152d897965c7a332c1504/lib/sbi/sbi_emulate_csr.c#L73

It seems fairly weird that this is a thing even, CSR reads are not supposed to slow down the trap handling much? (unless they are missing too, but then they are just variables on OpenSBI side anyways).
Plus OpenSBI is meant to be a reference implementation, but this clearly violates the spec:
[image]

SiFiveHolland, 2024-08-29:
I agree, and I sent a patch to fix this: http://lists.infradead.org/pipermail/opensbi/2024-August/007304.html

LekKit, 2024-08-29:
It might also make sense to delegate illegal instruction traps to supervisor mode if it's figured at runtime that no instruction emulation is necessary for the board in question (To speed up user->supervisor traps when necessary).
But this is of course a separate proposal. Maybe makes sense for another SBI call to ask for proper delegation?

SiFiveHolland, 2024-08-29:
How would you know that no instruction emulation is necessary? Right now that's not tracked anywhere; there is code to emulate everything, and if a platform doesn't need it, it just doesn't get called. And "necessary" depends on which ISA extensions you want to pretend the platform supports.

I think that would need a hardcoded list of "known-good" CPUs or SoCs in the firmware. An SBI call wouldn't be sufficient by itself, because S-mode software doesn't know which instructions can be emulated by a specific firmware version.

closed 2024-09-23 by avpatel