riscv-software-src/opensbi #372: Some ROM regions are overlapping

ddwolf, 2024-09-03:
here is the error log using [master.c4940a9](https://github.com/riscv-software-src/opensbi/commit/c4940a9517486413cd676fc8032bb55f9d4e2778) branch, after revert the opensbi code to v1.2, the error disappeared
```log
qemu-system-riscv64: Some ROM regions are overlapping
These ROM regions might have been loaded by direct user request or by default.
They could be BIOS/firmware images, a guest kernel, initrd or some other file loaded into guest memory.
Check whether you intended to load all this guest code, and whether it has been built to load to the correct addresses.
The following two regions overlap (in the memory address space):
fw_jump.elf ELF program header segment 1 (addresses 0x0000000000000000 - 0x0000000000023940)
mrom.reset (addresses 0x0000000000001000 - 0x0000000000001028)
```

SiFiveHolland, 2024-09-03:
Two possible fixes:
 1) Use `-bios fw_dynamic.bin` instead of `-bios fw_dynamic.elf`.
 2) Build OpenSBI with `FW_TEXT_START=0x80000000` (or apply [this QEMU patch](https://lore.kernel.org/qemu-devel/20240817002651.3209701-1-samuel.holland@sifive.com/) and choose any valid DRAM address).

alistair23, 2024-09-03:
Can you include the QEMU command line you are using?

ddwolf, 2024-09-04:
> Can you include the QEMU command line you are using?

```bash
qemu-system-riscv64 -m 2G -smp 16 -nographic -machine xxx-machine -bios none -drive if=...img -drive file=... -device pcie-root-port,port=0x10,chassis=1,id=root_port1
```