riscv-software-src/opensbi #376: How to build for QEMU/Spike?

Timmmm, 2024-10-02:
Hi, I'm trying to a build an OpenSBI ELF with Linux embedded in it so I can easily run it on QEMU, Spike and the Sail emulator.

I've built something, but running it in Spike it gets through the OpenSBI boot fine, but then seems to jump to the payload at 0x80200000 but then immediately hit an illegal instruction:

```
core   0: 0x00000000800004e6 (0x30200073) mret
core   0: >>>>  payload_bin
core   0: 0x0000000080200000 (0x10102464c457f) unknown
core   0: exception trap_illegal_instruction, epc 0x0000000080200000
core   0:           tval 0x00000000464c457f
core   0: >>>>  _trap_handler
```

The `vmlinux` entry point is

```
  Entry point address:               0xffffffff80000000
```

Can someone explain how the boot process is supposed to work? As I understand it:

1. Spike/QEMU/etc. loads your ELF into memory and a DTB also into memory somwhere, stores the DTB address in `a1`, the hart ID in `a0`, and then sets `pc` to the ELF entry point.
2. OpenSBI runs and sets up whatever machine mode things it needs. Then since the Linux kernel is already in the ELF and loaded into memory it can simply jump to Linux's entry point.

But it clearly isn't working.

`readelf` says `vmlinux` isn't position independent so surely it needs to be at the same addresses in the OpenSBI ELF and OpenSBI should be jumping to 0xffffffff80000000 not 0x0000000080200000?

If I look at the segments for `vmlinux` I get:

```
There are 9 program headers, starting at offset 64

Program Headers:
  Type           Offset             VirtAddr           PhysAddr
                 FileSiz            MemSiz              Flags  Align
  LOAD           0x0000000000001000 0xffffffff80000000 0x0000000000000000
                 0x00000000009d009a 0x00000000009d009a  R E    0x1000
  LOAD           0x0000000000a00000 0xffffffff80a00000 0x0000000000a00000
                 0x000000000004dae8 0x000000000004dae8  R E    0x200000
  LOAD           0x0000000000a4e000 0xffffffff80c00000 0x0000000000c00000
                 0x0000000000026880 0x00000000000268c8  RWE    0x1000
  LOAD           0x0000000000a75000 0xffffffff80c27000 0x0000000000c27000
                 0x0000000000012e08 0x0000000000012e08  RW     0x1000
  LOAD           0x0000000000a88000 0xffffffff80e00000 0x0000000000e00000
                 0x000000000034960c 0x000000000034960c  RW     0x1000
  LOAD           0x0000000000dd2000 0xffffffff81200000 0x0000000001200000
                 0x00000000000010b0 0x00000000000010b0  R      0x1000
  LOAD           0x0000000000dd4000 0xffffffff81400000 0x0000000001400000
                 0x0000000000115200 0x00000000001908b0  RW     0x1000
```

For the OpenSBI image I get:

```
  LOAD           0x0000000000001000 0x0000000080000000 0x0000000080000000
                 0x0000000000031bc8 0x0000000000031bc8  R E    0x1000
  LOAD           0x0000000000033000 0x0000000080040000 0x0000000080040000
                 0x0000000000002860 0x0000000000006020  RW     0x1000
  LOAD           0x0000000000036000 0x0000000080200000 0x0000000080200000
                 0x0000000001374f08 0x0000000001374f08  R E    0x1000
```

That last one is just 16 bytes more than the actual `vmlinux` file so I guess it has pretty much just dumped it there?

Any help on how this is supposed to work would be really appreciated!

Here is the Makefile I'm using so far (needs some work but basically I'm doing this):

```
# build Linux
make ARCH=riscv defconfig
make ARCH=riscv vmlinux
# build opensbi
make  FW_TEXT_START=0x80000000 FW_PAYLOAD=y FW_PAYLOAD_PATH=...../vmlinux PLATFORM=generic
# run in spike
spike .../build/platform/generic/firmware/fw_payload.elf
```

Full file:

```
.PHONY: download_gcc download_linux download_opensbi linux opensbi spike

RISCV_COMPILER := riscv64-lp64d--glibc--stable-2024.05-1
LINUX_VERSION := linux-6.11.1
OPENSBI_VERSION := 1.5.1

download_gcc:
	mkdir -p build
	curl -L https://toolchains.bootlin.com/downloads/releases/toolchains/riscv64-lp64d/tarballs/$(RISCV_COMPILER).tar.xz -o build/gcc.tar.xz
	tar -C build -xf build/gcc.tar.xz

download_linux:
	mkdir -p build
	curl -L https://cdn.kernel.org/pub/linux/kernel/v6.x/$(LINUX_VERSION).tar.xz -o build/linux.tar.xz
	tar -C build -xf build/linux.tar.xz

download_opensbi:
	mkdir -p build
	curl -L https://github.com/riscv-software-src/opensbi/archive/refs/tags/v$(OPENSBI_VERSION).tar.gz -o build/opensbi.tar.xz
	tar -C build -xf build/opensbi.tar.xz

CROSS_COMPILE := riscv64-linux-
export PATH := $(shell pwd)/build/$(RISCV_COMPILER)/bin:$(PATH)

linux:
	make -C build/$(LINUX_VERSION) ARCH=riscv CROSS_COMPILE=$(CROSS_COMPILE) defconfig
	make -C build/$(LINUX_VERSION) -j6 ARCH=riscv CROSS_COMPILE=$(CROSS_COMPILE) vmlinux

# FW_TEXT_START is 0 by default which doesn't leave space for the emulator bootloader.
# 0x80000000 is the default start of Spike's memory.
opensbi:
	make -C build/opensbi-$(OPENSBI_VERSION) FW_TEXT_START=0x80000000 FW_PAYLOAD=y FW_PAYLOAD_PATH=../$(LINUX_VERSION)/vmlinux CROSS_COMPILE=riscv64-linux- PLATFORM=generic

spike.dtb: spike.dts
	dtc spike.dts -o spike.dtb

spike:
	spike build/opensbi-$(OPENSBI_VERSION)/build/platform/generic/firmware/fw_payload.elf
```

SiFiveHolland, 2024-10-02:
You should use `make Image` for Linux and then build OpenSBI with `FW_PAYLOAD_PATH= ../$(LINUX_VERSION)/arch/riscv/boot/Image`. OpenSBI does not expect an ELF payload, but a flat binary following a specific calling convention (which the Linux "Image" format matches). For more information, see the documentation here: https://github.com/riscv-software-src/opensbi/tree/master/docs/firmware.

Timmmm, 2024-10-02:
Aha that worked, thank you!!

closed 2024-10-02 by Timmmm