riscv-software-src/opensbi #378: How to make Linux use OpenSBI for console output?

Timmmm, 2024-10-03:
Hi, sorry this isn't really an issue with OpenSBI but you were really helpful on the last issue so... :-D

I'm trying to make an ELF that will boot on Spike, QEMU and Sail and thanks to your help earlier I have one that works in Spike, and I *think* it's working in Sail too based on the instruction traces, except it doesn't print anything to the console.

Spike's DTS is like this:

```
...
  chosen {
    stdout-path = &SERIAL0;
    bootargs = "console=ttyS0 earlycon";
  };
...
  soc {
...
    SERIAL0: ns16550@10000000 {
      compatible = "ns16550a";
      clock-frequency = <10000000>;
      interrupt-parent = <&PLIC>;
      interrupts = <1>;
      reg = <0x0 0x10000000 0x0 0x100>;
      reg-shift = <0x0>;
      reg-io-width = <0x1>;
    };
  };
  htif {
    compatible = "ucb,htif0";
  };
};
```

Unfortunately Sail doesn't have a 16550 implementation, so I just removed it:

```
...
  chosen {
    bootargs = "console=ttyS0 earlycon";
  };
...
  soc {
...
  };
  htif {
    compatible = "ucb,htif0";
  };
};
```

If I use that then OpenSBI does successfully print to the console via HTIF, but Linux doesn't.

Do you know if there's a way to make Linux use OpenSBI for its console output? Does the SBI interface have support for console output actually? Maybe I just need to tell Linux to use HTIF too or something? I'm pretty sure this is possible because there's a very old Linux image in the Sail model repo that works (but it has other bugs). [Their DTS files](https://github.com/riscv/sail-riscv/blob/master/os-boot/rv64-2gb.dts) look very similar to mine. Hmmm...

Any help appreciated!

closed 2024-10-03 by Timmmm

avpatel, 2024-10-03:
Fixed a small typo in kernel parameters.

Timmmm, 2024-10-03:
Aha, with the typo fix it works!! :tada: 

Thanks!

Btw where is the documentation for these kernel parameters and the DTS and everything? (Assuming it exists!)