riscv-software-src/opensbi #382: OpenSBI doesn't build in a way suitable for spike

sfiedler0, 2024-12-01:
Currently I'm writing an own RISC-V kernel. 

I used the SBI in some parts of the kernel and previously ran QEMU together with OpenSBI's `fw_dynamic.bin`, and everything worked well. 

Now that I made my kernel do devicetree stuff, I recognized the QEMU devicetree being a bit weird (many empty strings in the strings block = many undefined names for properties of blocks).  
So I wanted to use spike, because… well… it's an official RISC-V simulator and it *could* provide a better device tree (at least I was able to directly dump the DTS)… 

My first idea was to just run the `fw_dynamic.bin` compiled with riscv64-linux-gnu toolchain and my `kernel.elf` together. After some time I understood spike needs only ELF files, so I used the `fw_dynamic.elf` file. It didn't work.  
After some research I found out spike doesn't like PIEs. However, I wasn't able to build the OpenSBI firmware without PIEs (I tried it once, with the newlib toolchain, PIEs disabled with patch [`2fe7528`](https://github.com/riscv-software-src/opensbi/commit/2fe7528c81453a713c6b946551e2d6f08df2d2d6), fw_payload with payload being a raw binary of my kernel and spike said `terminate called after throwing an instance of 'trap_store_access_fault'`).  
I, of course, set the `FW_TEXT_START` configuraton parameter to the value of `0x80000000`. 

Any ideas?

avpatel, 2024-12-01:
fw_dynamic is not appropriate for Spike because Spike does not generate and pass fw_dynamic_info structure.

Use fw_jump and fw_payload is more appropriate for Spike but since Spike loads ELF, you will have to use `FW_TEXT_START=0x80000000` when compiling OpenSBI.

sfiedler0, 2024-12-01:
Oops, builds with master, but not with v1.5.1…  
Now only thing left is getting my kernel to work in spike…

closed 2024-12-01 by sfiedler0