riscv-software-src/opensbi #385: Wrong boot CPUID on device tree header

IkerGalardi, 2025-01-08:
Currently the device tree passed to the supervisor software does not properly set the `boot_cpuid_phys` field in the header. It always returns 0 (I guess the default value set by QEMU in my case). The firmware should be able to modify this value so that software that relies in this field properly works.

The system I am currently using is `qemu-system-riscv64` with the `opensbi-dynamic.bin` version.

IkerGalardi, 2025-01-08:
I could contribute the fix but I don't really know the internal structure of the firmware. Are there any resources I could use? Where should this fix go (qemu specific or general)?

avpatel, 2025-01-08:
Any example of software which relies on boot_cpuid_phys ?