riscv-software-src/opensbi #394: Name OpenSBI already in use, up to the casing, since 2008

CCRDude, 2025-03-28:
Hi there,

Not trying to stir any trouble, but I was googling for one of our wiki pages and instead found lot's of references to the RISC-V Open Source Supervisor Binary Interface.

To sum it up: I'm the author of Spybot - Search & Destroy, an anti-malware solution existing since 1999. We published documentation of our open custom signature language called [OpenSBI](https://wiki.spybot.info/index.php/OpenSBI) somewhere around 2008, when our version 1.6 included support for it.

I wonder if no one did use a simple search engine before picking this name?

nc7s, 2025-03-28:
I wonder too: is it a registered trademark? Also, trademarks are registered under categories, so is _this_ OpenSBI registered under the category of _that_ OpenSBI?

If both answers are "no", then you are basically a John blaming another John for not using a simple search engine before picking this name John.

CCRDude, 2025-03-28:
You don't need to register trademarks to protect a name in most countries. Especially since I'm a John with a name that's part of a product with 600 million verified downloads, and many probably more that are anonymous, since we don't track downloads from our own servers, cover mounts, &c. Plus, I'm a John with at least 16 years in use, which is a strong argument in itself.

Actually though, I was simply starting the discussion out of curiosity, not for discussing the legal aspects of trademarks vs. registered trademarks, long-standing use, common trademark rights, &c. ;)

nc7s, 2025-03-28:
JFTR: "I wonder if no one did use a simple search engine before picking this name?" reads suspiciously like ridiculing things. Even if I'm ridiculing, it's reciprocal. (It's an accusation, after all. I'll not respond/reply if and why I can call my product Microsoft.)

CCRDude, 2025-03-28:
Well, I was really wondering. Whenever I pick a product name, I do spend a lot of time googling. Our company name, "Safer Networking", for example, didn't return a single search result back in 2003 in Google and AltaVista and Yahoo back then. Still, the trademark office rejected it as similarly common to "Safer Sex" - that was the official rejection notice ;)

You could see my question as ridiculing, or simply as curiosity and an invitation to say: "oh, it seems everyone thought someone else would do, but no one did".

If you felt offended or ridiculed (were you involved in the process?), I do apologize.